sarfata/kbox-firmware #6: Send PGN 127250, 127251 and 127257 based on IMU data

sarfata, 2016-06-28:
Respectively Magnetic heading, Rate of turn and Attitude.

See https://github.com/sarfata/kbox-firmware/wiki/Supported-PGNs

- 127250 and 127251 are supported by NMEA2000
- 127257 is not yet supported (see ttlappalainen/NMEA2000#10)

beaugunderson, 2016-08-13:
Noting that 127257 is now supported. :)

sarfata, 2016-10-15:
Creating new issues to keep track of the things that were not included here: Yaw (#23) and Rate of turn (#22).

sarfata, 2016-10-15:
Attitude and heading tested with canboat analyzer :white_check_mark: 

Magnetic heading tested on Raymarine i70: it works but incorrectly shows the heading as True.
Attitude tested on Raymarine i70: tilt is available. Roll and yaw do not seem to be supported.

closed 2016-10-15 by sarfata

lfrancois, 2017-09-25:
Why did you decide to remove this from the `develop` version of kbox? I guess it would be great to get as much data as possible on the n2k network.

sarfata, 2017-10-03:
@lfrancois it's back in #49

sarfata, 2017-10-03:
Actually, it's back in develop. In the new PR, it will also be sent via WiFi TCP (NMEA) and WebSocket (signalk).