sarfata/kbox-firmware #38: nmea sentence crashing the kbox

lfrancois, 2017-07-17:
It seems one of my NMEA 2000 device crashes the kbox. I am going to try to isolate the device to provide you more information.

Any idea how to troubleshoot it? I am running the `development` branch right now.

Here are the logs from the serial console.

```
WD main.cpp:90 Still running ... 0 connected clients - 36560 heap - IP: 0.0.0.0
WD main.cpp:90 Still running ... 0 connected clients - 36560 heap - IP: 0.0.0.0
D NMEA2000Service.cpp:92 Sending message on n2k bus - pgn=127508 prio=6 src=22 dst=255 len=8 result=success
D NMEA2000Service.cpp:95 TX: $PCDIN,01F214,00000000,16,000000FF7FFFFF01*57
D NMEA2000Service.cpp:92 Sending message on n2k bus - pgn=127508 prio=6 src=22 dst=255 len=8 result=success
D NMEA2000Service.cpp:95 TX: $PCDIN,01F214,00000000,16,010100FF7FFFFF01*57
D NMEA2000Service.cpp:92 Sending message on n2k bus - pgn=127508 prio=6 src=22 dst=255 len=8 result=success
D NMEA2000Service.cpp:95 TX: $PCDIN,01F214,00000000,16,FF0000FF7FFFFF01*57
D NMEA2000Service.cpp:92 Sending message on n2k bus - pgn=127508 prio=6 src=22 dst=255 len=8 result=success
D NMEA2000Service.cpp:95 TX: $PCDIN,01F214,00000000,16,FF4505FF7FFFFF01*53
D NMEA2000Service.cpp:37 Received N2K Message with pgn: 60928
```

lfrancois, 2017-07-17:
It seems not related to the PGN above. Same issue with a different PGN.

```D NMEA2000Service.cpp:37 Received N2K Message with pgn: 127505```

lfrancois, 2017-07-17:
After a quick troubleshooting session, it seems two of my NMEA2000 devices are causing the problem:
* Airmar P79
* NAVICO TEMPERATURE SENSOR NMEA 2000

lfrancois, 2017-07-19:
The issue looks to come from the SDCard logging. As soon as I comment it, it works as expected.
```
ludo@idefix:~/repos/kbox-firmware % git diff                                                                                            15:10 #6253
diff --git a/src/host/main.cpp b/src/host/main.cpp
index d59e5e5..1b11cb3 100644
--- a/src/host/main.cpp
+++ b/src/host/main.cpp
@@ -84,7 +84,7 @@ void setup() {
   SDCardTask *sdcardTask = new SDCardTask();
   reader1->connectTo(*sdcardTask);
   reader2->connectTo(*sdcardTask);
-  n2kService->connectTo(*sdcardTask);
+  //n2kService->connectTo(*sdcardTask);
 
 
   // Add all the tasks
ludo@idefix:~/repos/kbox-firmware %```

sarfata, 2018-06-14:
@lfrancois did you ever see this again? can we close this now?

lfrancois, 2018-06-15:
Since the new kbox, new firmware, new sd card, everything works fine! :)

closed 2018-06-15 by lfrancois