sarfata/kbox-firmware #42: kbox sensors data not available from 10110

lfrancois, 2017-08-20:
Hi @sarfata ! It seems the kbox stopped to forward its internal data, temperature, fake humidity, barometric pressure... Any idea how it could have happened? I am running the dev branch.

Please find enclosed some output from the kbox.

[kbox_logs_2017-08-20.txt.zip](https://github.com/sarfata/kbox-firmware/files/1236988/kbox_logs_2017-08-20.txt.zip)

sarfata, 2017-08-26:
Yeah I stopped sending it because I did not think it was very useful. Do you miss it? Can bring it back :)

What format would you like? PCDIN sentences or a NMEA sentence?

lfrancois, 2017-08-27:
PCDIN sentences works for me. Barometric pressure sounds useful. Do you get it from other devices? I need to check what else is covered in the code.

sarfata, 2017-10-03:
Solved in #49

closed 2017-11-06 by sarfata