sarfata/kbox-firmware #43: NMEA0183 outputs?

ronzeiller, 2017-08-29:
You write:
Two NMEA0183 outputs

How to activate NMEA0183 output?

sarfata, 2017-09-01:
Hi,

They are not used in the code yet but you can just write to Serial2 and Serial3. What information would you like to see on those outputs? Tell me more and I will help!

thomas

ronzeiller, 2017-09-02:
Hi Thomas,

All informations available, which can be described as pure NMEA0183 could be written to NMEA1 or NMEA2. 
(The $PCDIN sentences would probably make no sense on Serial Output, as not understood from any navigation program directly connected)
This would make KBox more universatile in its use.
For examle is this needed to connect KBox to a Raymarine´s (formerly Tacktick) Micronet Interface Transmitter for communication with Racemaster and/or other Micronet Displays.

Thank you!
Ronnie

CaptainRon47, 2017-11-27:
My specific interest here is to be able to set way points from OpenCPN and send them to my GPS.  But my GPS only has one input (at 4800) and so I cannot send it both input instrument data and waypoint data separately.  Currently my GPS, at the helm, receives all my instrument NMEA data from my NMEA instrument interface box as it is also sent to the KBox (my NMEA interface box can source to both the GPS and Kbox simultaneously).  I would like to be able to (instead) repeat all my instrument data and add way point sentences from OpenCPN (on my Rpi) as it returns to the Kbox and comes from the NMEA outputs on the Kbox.  So KBox would have to act like a true multiplexer by both repeating the input data and adding data coming from OpenCPN (or any other navigation program).

sarfata, 2017-11-27:
I think we can make a dumb version of output to NMEA work easily but we are going to run into issues when there is too much data flowing in KBox than what we can send at 4800 bauds. In this case, some messages will get lost and we will need some kind of filtering, or reduction of speed to make sure this does not happen. We can implement this later.

sarfata, 2017-11-27:
I think we can make a dumb version of output to NMEA work easily but we are going to run into issues when there is more data flowing in KBox than we can send at 4800 bauds. In this case, some messages will get lost and we will need some kind of filtering, or reduction of speed to make sure this does not happen. We can implement this later.

CaptainRon47, 2017-11-28:
Thomas,
I don't need to send all NMEA 0183 data out on NMEA1_output, only what my GPS ( and perhaps my AP) uses, because those are the only devices connected to it.  This would be boat instrument data; such as; paddle wheel speed, depth, (maybe wind speed and direction) plus sentences coming from OpenCPN like waypoint settings and goto directions.

sarfata, 2017-12-01:
Ron, how would you prefer to configure this?

1. List the data that you want to send (speed in water, depth, wind speed and direction), which might be a list of signalk keys.
2. List the sentences that you want to generate (DBT for depth, MWV for wind, etc)?

What would be best in your opinion?

ronzeiller, 2017-12-02:
Apart of your two options I think we need a third: 
On which output the sentence should go? (The fast NMEA_1 or the slow NMEA_2 with limited amount of sentences, and/or WiFi) (That would mean that the output should probably be added to the message list....)

I think that we should stick to the kind of data format which is sent to (or received from) the interface.
That means if NMEA0183 is concerned then we should speak in Depth (DBT), Wind (MWV) etc..
(additionally later on, someone could select deprecated sentences too, if there is a demand for it)

[When a friend of mine and I started our project (before I was switching to KBox) we even thought of the possibility to let the customer configure each "Talker-ID" to stay compatible with all sorts of navigation programs. May be this is something for keeping in the back yard of our brain... :-)]

If the user chooses Signal K on the output interface (just makes sense on WiFi at the moment?) then a list of Signal K keys would be needed.

sarfata, 2017-12-28:
closed by #107 

You can now enable the output and configure what KBox should send on the output. Note that only the data parsed by KBox can be sent to the output. This is a bit limiting right now but we will extend it. See [Configuring KBox](https://github.com/sarfata/kbox-firmware/wiki/Configuring-KBox) for ore details on what sentences can be output at the moment.

closed 2017-12-28 by sarfata