sarfata/kbox-firmware #45: WiFi connection closed by foreign host

ronzeiller, 2017-09-18:
Hi Thomas,

Datas are coming over WiFi and all is working nice for a while.
No it says "Connection closed by foreign host."

Sometimes early, just now after 2 hours.

Did you experience similiar?
Any suggestions to make the webserver more stable?

Thanks,
Best Regards
Ronnie

sarfata, 2017-09-24:
Can you give me more information on the program you are using as a client? And what branch are you testing? Master or develop?

I have seen some problem with develop but not with master.

ronzeiller, 2017-09-25:
I am using the master branch on a Teensy 3.6 with NodeMCU ESP-12 V2.0
New ADC Library and edited platformio.ini for working with T3.6, everything else from your code.

Are there the same ESPAsyncTCP libraries used for master and develop or anything else changed concerning the Webserver itself, beneath all the KSignal things in develop?

I did not explore the code that much for the WiFi yet, may be there exist a more stable library version for the Webserver?

sarfata, 2017-09-25:
Thanks. What client are you using? How do you connect to it? iNavX? Some other app? Have you tried from a computer?
Can you reconnect immediately?

First thing I would do if I were you is try to see if the ESP is rebooting. You could have a debug message with the uptime sent every 10sec on the tcp connection, or on the serial port.

This way you would know if the connection lost is due to a reboot or some other issue. That will be very helpful.

sarfata, 2017-09-25:
ps: I am glad to hear KBox code is running on a Teensy 3.6. I did not get time to try this yet!

ronzeiller, 2017-09-25:
Aaah, sorry, forgot to mention:
Working on a Mac here, client is telnet in terminal.
Telnet is stopping with: "Connection closed by foreign host."
But WiFiTask is still running, telling me "Still running..."
Immediate reconnect possible. 

With openCPN and the NMEA-Debug Window. Sometimes it stops, but starts again by itself after around 8sec....
Could swear, that it was completely stopping, as telnet does, but now it runs quite stable.

By the way, great work, your software!

ronzeiller, 2017-09-25:
Seems that I was happy too early...
Just tested two computers connected to the same WiFi, both openCPN runnng.
Mac: openCPN 4.4.0 -> Datas are coming ok and immediate responding to changes on the input

Windows 7: openCPN 4.6.0 -> **Datas are coming around 1 minute later!!** (damping in preferences all zero.)

ronzeiller, 2017-11-05:
If I understood it right, Teensy has asynchronous writing to serial.
As you suggested in Issue "NMEA Outputs" I just did Serial.print for output of NMEA Sentences (for every parsed and converted N2k-PGN)  to Serial.

Now I am asking myself, about the advantages of the loop and the LinkedList "sendQueue" in the WiFiTask? (May be I am missing some of the advantages of the Visitor pattern....)
Would you suggest to do a similar thing in the NmeaReaderTask for writing to Serial?

BTW, the "leftoverToSend" String (which could grow endless?)  in WiFiTask seems critical to me, as it could lead to time delayed datas and was probably the reason for the delay I saw with openCPN on windows?

sarfata, 2017-11-06:
Hey @ronzeiller, sorry for the delay here. Are you still seeing the issue on Windows? Is the connection delayed by 1 minute? or the data? What if you try a telnet from windows?

You are asking very very good questions :) Let me try to explain:

* Teensy has a write buffer for the connection to the WiFi module which [by default is 64 bytes](https://github.com/PaulStoffregen/cores/blob/master/teensy3/serial1.c#L40). In the master branch we do not change this but [in KBox `develop` branch it is configured to 512 bytes](https://github.com/sarfata/kbox-firmware/blob/develop/platformio.ini#L15). For the NMEA1 and NMEA2 serial ports, all versions of KBox currently use the [default value of 40 bytes](https://github.com/PaulStoffregen/cores/blob/master/teensy3/serial2.c).
* When we write to the serial ports, the data is pushed to this buffer if there is enough space. If there is no space left, all or some of the data will be lost. That is because [we have defined a timeout of 0ms (no timeout)](https://github.com/sarfata/kbox-firmware/blob/develop/src/host/main.cpp#L118).
* So yes it feels like an asynchronous serial port as long as you keep in mind that the buffer might be full and in this case the write will return without pushing all the data.
* In the master branch, the sendQueue is used to buffer the data and make sure we do not send it too fast but it could grow out of control and exhaust the memory.
  * The loop executes really fast (over 100 times per second) so I doubt this is the cause of the delay on Windows.
* In the `develop` branch we push the data to the write buffer and if there is not enough space the data will be lost (but we will not crash).

Now to go back to your specific situation:
 * First, I recommend working from the develop branch. It is much better than the master branch in many ways and will soon be replaced by master.
 * You can just print the data to the serial port but any sentence longer than 40 bytes would be "cut" (and therefore invalid). Also when multiple sentences are generated at the same time or very close to each other, you will also lose some data. 
 * To avoid this, the simplest solution will be to increase the size of the Serial2 (or Serial3) write buffer in `platformio.ini`. This is what I would recommend here.

Keep in mind that Serial output at 38400 or even 4800 will always be slower than NMEA2000 can be so dropping sentences is required here.

sarfata, 2017-11-23:
Do you still have that problem @ronzeiller? Did you figure out what was going on?

ronzeiller, 2017-11-24:
No, I don't have any idea was it was.
With the develop branch it seems to work now in openCPN on Mac. 
But there are not so many PGNs translated yet in develop branch.

I will redo the test with the two computers, comparing the values from serial output and WiFi, but we need some more NMEA0183 for writing (as I had on my test code from master branch)

May be better to keep this issue open ;-)

ronzeiller, 2018-02-17:
Close issue, because of in the meantime completely outdated KBox version.

closed 2018-02-17 by ronzeiller