sarfata/kbox-firmware #50: After rebuild of KBox-firmware KBox runs only in test-mode

Heertsch, 2017-10-12:
Hi Thomas
perhaps a very beginer question: I changed NMEA port 1 to 4800Bd and integrated KBox WiFi into my network.
Then I compiled and uploaded the firmware. All runs fine, but after Reset KBox shows only the testsuite.
Do I have to push some button?
Thank you
Andreas

sarfata, 2017-10-13:
Looks like you uploaded the wrong firmware. The test firmware is uploaded when you select the target `mfgtest`.

Make sure you select the firmware called `host` first (for the main cpu) and then the firmware called `esp` for the WiFi module. When both are programmed, you should have everything working as expected.

sarfata, 2017-10-13:
Are you using the command line or the Atom IDE? If you are not sure how to change target, please refer to the "Getting Started Guide" or give me a few more details so I can help you.

Heertsch, 2017-10-13:
I got it: after 1st upload platformio on Atom request again a reset of kbox and then it loads some more. the 3rd request of reset loads the test software. If I restart the upload and push reset only once, it runs in normal mode.
But there seems to be still some problems: I see in the serial-terminal, that there is no IP (0.0.0.0). I can connect with the KBox AP. but terminal debug states 0 clients.

Then I changed the baud rate to 4800 on NMEA183 (both ports) and sent some wind-signals to the port 1 (- and GND connected), but no result in the 2nd KBox window.
Do you have some hints?
Thank you
Andreas

sarfata, 2017-10-14:
@Heertsch when you are programming the box, you can choose exactly which of the modules you want to program. There is a screenshot in the developer setup:
https://camo.githubusercontent.com/ab31515b3c3848e615b77e38b245f9c6e3908104/687474703a2f2f692e696d6775722e636f6d2f675672623066492e706e67

This way you can choose host or esp and avoid having to guess which request is currently in progress.

If KBox is configured as an AP, it will display 0.0.0.0. The actual IP when configured as an AP is 192.168.4.1. If you take iNavX or another software and configure it to connect to port 10110 of this IP, you should see "1 connected client". You should also start receiving your wind signals frame via the TCP connection.

Another way to make sure the wind frames are received is to press the rotating button once to switch to the stats page.

Heertsch, 2017-10-16:
Bingo! 
- I did'nt expect, that the debug output of IP 0.0.0.0 is wrong.
- I used iNavx as Android version - and there is no implementation of TCP. (Now I bought iNavx for iOS and voillà: it works even with own IP-address; set with WiFi.config())
- The NMEA input still does not work. (Stats page: NMEA rx1 =0) But I will test it further....
Thank you!
Andreas

Heertsch, 2017-10-23:
WiFi runs as excpected (even with `WiFi.begin()` and `WiFi.config()` displaying `WiFi.localIP()`)
But with `NMEAInput `I have no luck. on both chanels no reaction (also not on the result page `N1 Rx: 0` `N1Rerr 0`. connected is NASA Wireless Wind with 
`GND` at `NMEA in1 -` 
`NMEA` at `NMEA in1 +` 
 `// Initialize serialports`
  `digitalWrite(nmea1_out_enable, 0);`
  `digitalWrite(nmea2_out_enable, 0);`
  `pinMode(nmea1_out_enable, OUTPUT);`
  `pinMode(nmea2_out_enable, OUTPUT);`
  `NMEA1_SERIAL.begin(4800);`
  `NMEA2_SERIAL.begin(4800);`
It seams, that the serialEvent is not fired

sarfata, 2017-10-24:
Hi @Heertsch,

I could not find a manual online for the NASA wireless Wind. If it only has one NMEA signal output (which seems to be the case based on your description), you should only connect this to NMEA In1+. You can leave the NMEA In1- disconnected.

The grounds of both will be connected via the boat's ground.

Have you checked that the NASA wind actually sends anything? Do you know it is working for sure?

Finally, what version/branch of KBox are you running? From master? or from develop?

thomas

Heertsch, 2017-10-24:
Bingo! Next step reached: I omit the GND connection and KBox is counting NEMA input.
Next challenge: Forwarding the MWV-record to WiFi. (There are no records forwarded.) 
Does KBox forward all kind of records?
Does KBox also forward records from WiFi to NMEA-output? I think about a steering of the rudder with commands started by mobile for better manege of single handed arrival in harbors
> Finally, what version/branch of KBox are you running? From master? or from develop?

I suppose I use the master version of KBox software.
Andreas

sarfata, 2017-10-27:
Hi @Heertsch 

You should not have to do anything to see the data forwarded to port 10110.

Do you see anything coming in on port 10110? (besides the lack of MWV - is there anything else?)

Can you check on the SDCard in one of the kbox logfiles to see if the data is there?

KBox does not forward records from WiFi to NMEA output yet but many people have asked for this and I will add this ability. 

If you try the `develop` branch you will find that it is more modern and has more features but it is also a little more experimental :)

Heertsch, 2017-10-28:
Next bingo: KBox delivers NMEA183 and NMEA2K records as expected. (NMEA2k CAN-bus works only with resistors connected on the data wires.)
I have still questions, but I think home for such discussions is https://groups.google.com/forum/#!forum/kbox-discussion
so I think we change the thread to this board.
Andreas

closed 2017-10-28 by Heertsch