sarfata/kbox-firmware #51: Output data in NMEA format via serial ports

sarfata, 2017-10-20:
Multiple users have requested that KBox should relay available data to its serial ports.

I am thinking of the following:
* NMEA1 - Send everything we have at 38400
* NMEA2 - Send only GPS data at 4800

Data would be generated from internal SignalK data. This means anything coming from one of the NMEA port, the NMEA2000 or WiFi connection would be repeated as long as KBox understands it and knows how to translate it to NMEA.

ronzeiller, 2017-11-05:
Generally spoken KBox is a very nice hardware to do a lot of things with boat datas.
The firmware (master branch) is also very good which allows a lot of possibilities.

What is really missing from my point of view is a concept for flexible configuring KBox for all the different usabilities. 
So IMHO the output on its serial ports should be much more flexible as a user might want to:

- use KBox configured as an Actisense NGT-1 (output of binary Actisense format) to Expedition
- feed NMEA0183 sentences to openCPN
- send calibrated/calculated datas back to NMEA2000-Bus to use boat displays
- send NMEA0183 by WiFi to the iPad  ragatta or navigation software
- get "real" calculated values for True Wind
- get information on current
- send SignalK to ...? 
- have a mixture of all this above.
- or, someone might have a good Compass system and will not relay on the built in BNO055 for HDG  (just want to get Heel/Pitch from the internal sensor)

So I think serial output should be much more flexible, also in data rates, as it depends if you need all sentences from the N2k-Bus up to 4Hz and more, or just for displays, eventually nicely smoothed by a filter in 1 Hz.

Personally I like the original KBox Firmware much more than the KSignal based "develop branch", as IMHO this is just another kind of proprietary output data format for a very special kind of use.....
(KBox vs. iKommunicate?)

But to come back to the topic: May be you should define clear usages of KBox first and then decide what, how often and where to send datas over WiFi or serial ports?

sarfata, 2017-11-06:
I agree with you that it is easy to quickly come up with lots of combinations ;) So far I have been trying to be very pragmatic and only solve real problems from real boats. I am happy to help you in the same way. Which one of these scenarios apply to you? What do you need?

Every time I learn a little bit more about actual problems in the real world and this helps me plan a future web-based configuration interface.

Also, I want to point out that the `develop` version should do everything the `master` version does. SignalK is a bonus but it does not remove other feature.

ronzeiller, 2017-11-06:
I agree that it doesn't make sense to develop on the "old" `master` branch. 
Just started with that, because it seemed more simple for me to understand the code.

I did some things which might also be of interest for you, like parsing NMEA2000 sentences, having an ini-file with settings to change things on the fly, date & time from GPS for new logging file for each day.
So may be better to go back to the start and continue to work on the `develop` branch. 
(Started with a pull request for Teensy 3.6)
(At the beginning of our work I wanted to buy a KBox, but then I thought may be I am more flexible with the Teensy 3.6....)

## The scenario for us will be:
## INPUTS: 
**NMEA2000**
**Serial2:** GPS with uBlox8 (which seems more accurate than boat's GPS, especially at low speed. Aim is to calculate current more accurate)

## OUTPUTS: 
**USB Option 1:** Actisense Binary for Expedition (instead of Actisense NGT-1)
**USB Option 2:** NMEA0183 for openCPN with tactics_pi for Performance analysis
**WiFi / UDP:** NMEA0183 for iPad Navigation (just the interesting sentences like Wind, Boatspeed, SOG/COG, Depth, Position)

**Optional Output, Serial 4800 Baud:** for additional Tacktick Micronet Displays (just few sentences for display only, may be smoothed by IIR-filter) (in the future own affordable  jumbo displays would be nice to have....)

**SD-Card Logging:** 
all sentences at the beginning of development, later on, smoothed datas in a format capable for replays

**The background is:** 
we are doing quite a lot of regattas on chartered (one design fleet) boats with very few time for training and basic calibration of the instruments.
We want to have a "box" connected to NMEA2000 Bus, getting datas for navigation and (own) performance maths, writing back calculated "corrected" datas e.g. boat speed and True Wind to the boat's MFDs.

That's all :-)

sarfata, 2017-11-23:
Closing this ticket which is a duplicate of #43, opened earlier.

closed 2017-11-23 by sarfata