sarfata/kbox-firmware #53: Keep track of total distance sailed by boat

sarfata, 2017-11-06:
[Suggested by Andreas Heertsch](https://groups.google.com/d/msgid/kbox-discussion/27cba8f4-32a3-49a9-9e5b-6d21f59835a3%40googlegroups.com?utm_medium=email&utm_source=footer)

> It would be fine, if KBox has a logging-function for the accumulated distances the boat has driven. KBox would be the best place, to accumulate (and backup this value on SD-card),  because KBox is always running. (If I use an old Smartphone for logging, I am afraid, that miles are lost, if the Smartphone is in "bad mood").

ronzeiller, 2017-11-06:
may be (also) stored to the  EEPROM as the SD-Card could be changed?
(e.g. as 10 minutes Task or when log changes more than nn miles) 

BTW does the EEPROM get harmed by writing to it too often? 
(Thinking also of storing some basic ini-settings to the EEPROM.....)

sarfata, 2017-11-23:
> BTW does the EEPROM get harmed by writing to it too often?

Great question!

On the Teensy the EEPROM is not really an EEPROM. It's a "fake EEPROM" in RAM which is persisted to flash by a special peripheral in the micro-controller.

In the datasheet for the [teensy 3.6 micro-controller](https://www.nxp.com/docs/en/data-sheet/K66P144M180SF5V2.pdf) (3.2 is similar) the interesting part about this is in section 3.4.1.4.

I could not find exactly what is the " EEPROM backup to FlexRAM ratio" used by Teensy but you would get at least 400,000 writes which is about 10 years. If you made a system that does not always write in the same place but instead pushes to different memory location, you could quickly get an even larger number of writes.

So yes it's possible but it needs to be looked at carefully.

sarfata, 2017-11-23:
I think the easiest way to do this if we decide to store the log value in this "EEPROM" is just to make sure we use a "backup to FlexRAM ration" of at least 128 which gives us a minimal number of writes of 1.26 million. That is 23 years at one write per 10 minute.