sarfata/kbox-firmware #57: Library for global variables for Boat Datas, API for Plugins

ronzeiller, 2017-11-07:
At the moment we have a lot of different KMessage Types, KSignal-Strings, Visitors with LinkedLists,  and so on.

Would't it be nice to have something like global variables or a class just for actual boat Datas for Apparent Wind (Angle, Speed, Direction), True Wind (Angle, Speed, Direction), Depth, COG, SOG, Heel and so on which we can write to and get the most actual data where- and whenever we need them?

May be as a (basic, central) library so users could extend the class in some kind of plugIn to whatsoever they need? (Similar as openCPN....)

KBox Firmware could become an even more flexible framework for doing a lot of interesting things on a boat.
E.g. most of that is already possible by the hardware: http://www.yachtd.com/products/
But also thinking of future displays getting datas by Bluetooth (similar to Tacktick) or KBox doing things like B&Gs Hydra or Sailmon does and so on....

sarfata, 2017-11-07:
I completely agree with you and this is why the develop branch has the `SKUpdate` object with variables to describe everything going on in the boat. I chose to follow the SignalK schema because they already spent a lot of time defining it and it should be exhaustive.

See https://github.com/sarfata/kbox-firmware/blob/develop/src/common/signalk/SKUpdateSyntacticSugar.generated.h

Because updating this file is tedious, there is a script (`sk-code-generator.py`) to do it from a list of keys we want to support: https://github.com/sarfata/kbox-firmware/blob/develop/src/common/signalk/signalk.json.

If you need new variables, just pick them from the SignalK standard, add them to this file and re-generate the other files. This is how I am currently adding wind support for example. I will push my branch so you can take a look.

The KMessage stuff is left-over from the master branch and should be completely replaced with this new format.

ronzeiller, 2017-11-08:
The SKUpdate object Looks fine.

But, if I understood the system, we would need  it as library to get and set values by another library which would then act like a plugin to KBox.

Then we could implement things without touching your src code.
Switching on a plug-in could be done in ini- or settings file or simple #define

It is just a vague idea, may be there are much better solutions for something like that...

sarfata, 2017-11-08:
I agree that we should make this a library one day. I have not done that yet because I wanted to use it long enough to make sure the API is clean and stable before starting to encourage other people to use it. 

I do not think we are there yet and so for now it's just easier to work on it here directly (I think).

ronzeiller, 2017-11-08:
Totally agreed, that this is something for the future. 
Great, that you think similar in this matter and that you will have this in mind when you do further developments.

closed 2017-11-27 by ronzeiller