sarfata/kbox-firmware #62: As a user, I want to configure the speed of NMEA input/output

sarfata, 2017-11-23:
Users should be able to configure the speed of NMEA ports without recompiling the firmware.

CaptainRon47, 2017-11-28:
Thomas,
I have my NMEA interface box connected to NMEA1 and it runs at 4800, while I have my radio AIS output connected to NMEA2 and it runs at 38400.  So I have them set as such.  Once they are set I don't see a need to change them, so the request to set them without recompiling is not mine.

sarfata, 2017-12-01:
Thanks for the precision Ron. I do think that this is one of the most basic configuration and people should not have to recompile for that so I hope to have a configuration file on the sdcard that defines this.

sarfata, 2017-12-28:
closed by #107

closed 2017-12-28 by sarfata