sarfata/kbox-firmware #64: As a user, I want to configure the orientation of KBox in my boat

sarfata, 2017-11-23:
See work by @ronzeiller here:
https://github.com/ronzeiller/kbox-firmware-teensy36/commit/e5ba6a3ce79cb01102524898e0d53f846a6e21e7

ronzeiller, 2017-11-24:
What do you think
Is it better to do with the setting on axis or to calculate the Euler as you did it?
I think we should not mix up the methods and find a low cpu consuming way.

sarfata, 2017-11-27:
I asked myself the same question but I did not see any benefit to make the change in the axis. We probably still need some post-processing so I think it's better to do everything in the driver because it is easier to read and understand than changing the axis.

The only exception would be if we wanted to support KBox flat on the ground or the ceiling, or installed sideways but nobody has mentioned this yet. I don't think we need to support this now.

ronzeiller, 2017-11-27:
> I asked myself the same question but I did not see any benefit to make the change in the axis. We probably still need some post-processing so I think it's better to do everything in the driver because it is easier to read and understand than changing the axis.

driver?
Do you mean the library or the IMUService?

> The only exception would be if we wanted to support KBox flat on the ground or the ceiling, or installed sideways but nobody has mentioned this yet. I don't think we need to support this now.

I have to put my KBox flat with the left side to the bow....

sarfata, 2017-11-28:
Driver: I meant the library. That is what I call the driver. Sorry.

> I have to put my KBox flat with the left side to the bow....

Ah ah ah. Making things hard on purpose! Let's do some testing to see what is required to support it then.