sarfata/kbox-firmware #74: Save accelerometer calibration and restore it when rebooting

sarfata, 2017-11-23:


ronzeiller, 2017-11-24:
I have made a routine for this, I will make a PR, have to look for it.

It did not help me to get proper values, but I have the box on my desktop and without moving the sensor the values for HDG are rather useless and wrong...

Heel and Pitch are reliable good and not drifting.
On one of my versions I had a long button press for storing offset values for heel/pitch to "calibrate" the boat in the marina, but it was for the master branch ;-)
To load those values from memory at rebooting is nice to have (especially when rebooting is necessary during sailing....)

sarfata, 2017-11-27:
Can you be more specific about "they are rather useless and wrong"? Did you connect the same sensor to your teensy? Is the value off when you compare it to a magnetic compass? Or completely bogus and jumping all over?

This sensor is supposed to be really good and is used in drones and many other applications. I think we can make it work very well. When I take the time to calibrate it at home (following the procedure in the datasheet), it gets very precise afterwards. I have not tried to calibrate and compare to a navigation compass on a boat yet.

sarfata, 2018-02-17:
I think we can close this ticket now. I agree we are not very happy with the result but the calibration is saved and restored.

closed 2018-02-17 by sarfata