sarfata/kbox-firmware #75: As a user, I should be able to restart compass calibration once KBox is installed on my boat

sarfata, 2017-11-23:
Most users will never need to recalibrate accel/gyro but we should have a way to re-start the magnetic calibration once KBox is installed in the boat.

ronzeiller, 2018-01-02:
@sarfata 
If I understood the mechanism of the BNO055 then:

- the sensor always needs calibration **after each power on** -> this means that you just have to move the KBox for calibration (there is no restart)
- it is self calibrating all the time when powered on, but therefor the sensor has to be moved! 
- it **does not make sense to store any calibration value** to the EEPROM and recall it during powering on. A recall of calibration would just recall the last values, which means the same value for heading would appear, but when the boat was moving in between (= sensor has moved) this heading would be wrong. (so the heading would recover to a true value by itself but therefore the sensor has to move. If the sensor does not move it will stay wrong)

sarfata, 2018-01-03:
@ronzeiller I don't think I agree. I have seen many other projects where they store/restore calibration values. I think the idea is that it allows the sensor to start faster and get to a calibration status much faster.

ronzeiller, 2018-01-03:
I was reading a lot of forums about the BNO055 but I could not find any other approved successful  project for boating. 

Anyway, now we also have store/restore calibration values in KBox, let's see if it improves the calibration process.

sarfata, 2018-01-08:
My idea with this ticket was to have a way to go back to the '0' calibration that the sensor starts with but since it is always calibrating, I don't think this is actually going to be helpful at all and I will close this.

closed 2018-01-08 by sarfata