sarfata/kbox-firmware #85: NMEA2000 Parsing, missing actions, questions....

ronzeiller, 2017-11-29:
## Working list for SKNMEA2000Parser

### PGN 128259  Boat speed
see discussion with Timo: https://github.com/ttlappalainen/NMEA2000/issues/84
IMHO we should not store ground speed there, as it will probably not coming and some microseconds later we get ground speed by GPS.

### PGN 126992  System Time Date UTC
I do not know how to make an entry in signalk.json for Signal K fields

### 127250 Vessel Heading

> Official NMEA2000 doc says:
> Heading sensor value with a flag for True or Magnetic. If the sensor value is Magnetic, the deviation field can be used to produce
> a Magnetic heading, and the variation field can be used to correct the Magnetic heading to produce a True heading.

So we could calculate True heading for output if values for variation AND deviation are there.

### Attitude
Would like to add Attitude to SKNMEA2000Parser, to switch from internal sensor to boats values for heel and/or pitch.
Selectable by config setting.
OK for you?

### Config setting needed for
Seasmart Output vs. NMEA0183 Output 
As we soon have a lot of NMEA0183 sentences for navigation
Each NMEA sentence should have its own config-selector for en-/disable

sarfata, 2017-12-01:
> ### PGN 128259  Boat speed
> see discussion with Timo: https://github.com/ttlappalainen/NMEA2000/issues/84
> IMHO we should not store ground speed there, as it will probably not coming and some microseconds later we get ground speed by GPS.

Ok.

> ### PGN 126992  System Time Date UTC
> I do not know how to make an entry in signalk.json for Signal K fields

You need to add something like: 
```
{
  "path": "environment.time.timestamp", 
  "type": "stringValue",
  "description": "A time reference for the vessel. All clocks on the vessel dispaying local time should use the timezone offset here. If a timezoneRegion is supplied the timezone must also be supplied. If timezoneRegion is supplied that should be displayed by UIs in preference to simply timezone. ie 12:05 (Europe/London) should be displayed in preference to 12:05 (UTC+01:00)",
}
```
(from https://github.com/SignalK/specification/blob/master/schemas/groups/environment.json#L346)

The problem here is that the generator does not support `stringValue` types yet so you probably want me to do this for you.

> ### 127250 Vessel Heading
>
>
> So we could calculate True heading for output if values for variation AND deviation are there.

I think the converters should just translate what they receive as precisely as possible to signalk and not do any interpretation. We can then have modules (aka the "Data Computers") that will process data and generate more values if we want.

> ### Attitude
> Would like to add Attitude to SKNMEA2000Parser, to switch from internal sensor to boats values for heel and/or pitch.
> Selectable by config setting.
> Ditto for Magnetic Heading, if it is on the N2k-bus than I would rather like to switch to this source. 
> OK for you?

Completely ok. Makes a lot of sense.

> ### Config setting needed for
> Seasmart Output vs. NMEA0183 Output 
> As we soon have a lot of NMEA0183 sentences for navigation
> Each NMEA sentence should have its own config-selector for en-/disable

Agreed too.

ronzeiller, 2017-12-19:
@sarfata
Nice overview to see which NMEA0183 can be written from a N2k PGN (and which PGN information else is needed) found at Actisense:

http://www.actisense.com/wp-content/uploads/2017/07/2014-04-24-NGW-1-AIS-Conversion-List-2.420.pdf

ronzeiller, 2018-01-05:
@sarfata
Could you please advise how to:
add NMEA2000Config to SKNMEA2000Parser --> for enable/disable incoming PGN's

I would like to do some calculations in the SKNMEA2000Parser before it comes to an update. Starting with boat speed and leeway (which has it's own config). What would be the best way to integrate a class for doing that?

I have made a rough config to give you an idea:
```
struct PerformanceConfig {
  bool enabled;
  String boatSpeedCorrectionTable;
  int hullFactor;         // usually between 8....10, here multiplied by 10
};
```
Thank You!

sarfata, 2018-01-08:
@ronzeiller I have not seen what you have in your performance PR but I think the best way would be to make a class that subscribes to `SKHub` and will receive every signalk update. If you see an update with the stuff you need, then you can generate a new update and send it as well.

There is no way right now to modify an update while it's going through the system. That will require some more work and if that's what you need to do, then it might be best to do it in the NMEA2000Service. It's not ideal because it will only work for data coming from NMEA2000.

ronzeiller, 2018-01-08:
Correction of performance PR: #117

ronzeiller, 2018-02-09:
> @ronzeiller I have not seen what you have in your performance PR but I think the best way would be to make a class that subscribes to SKHub and will receive every signalk update. If you see an update with the stuff you need, then you can generate a new update and send it as well.

@sarfata
Done as you suggested. 
Also put new SKSource "PerformanceCalc" so the output to WiFi, Serial, N2k, SD-Card could be chosen "corrected/calculated"  by config setting.