sarfata/kbox-firmware #87: Voltage Readings

ronzeiller, 2017-12-02:
Thomas,
I was trying now for hours to find out why the voltage reading in "my T36" KBox is not working, but without result.

The strange thing is, I do not see any difference between master-branch and develop-branch.
In master-branch it is working and in develop I get 0.0V

If I do
` DEBUG("analogRead: %i",analogRead(supply_analog));`
it gives me a reading of around 2030 in master (which calculates with the voltage divider to correct supply voltage.)
and 
`D ADCService.cpp:35 analogRead: 6`
in develop-branch, as if the `adc.setResolution(12);` is not working.

May be you have an idea, what it is?

ronzeiller, 2017-12-02:
Sorry, it was a stupid mistake!!!
In my Box I have CAN Standby on the same pin as Supply battery, so the pulldown made 0V
(today I changed pins to become compatible with original KBox again and I forgot the CAN.... :shit:)

ronzeiller, 2017-12-02:
Just found out, that it is not possible to change the pin, because on the T3.6 Pin A14 = pin 33

Would you mind if we put into the `board.c `something like:
```
/* CAN Transceiver */
#if defined (__MK66FX1M0__)
  static const pin_t can_standby = 32;
#else
  static const pin_t can_standby = 33;
#endif
```
Just for the next 100s of PRs? :-)
Thank You!

sarfata, 2017-12-04:
No problem. I think the best way is to use a `#ifdef BOARD_RONZEILLER` (or something like this) and to add a `-DBOARD_RONZEILLER` in the build flags of your environment (in platformio.ini).

Two reasons:
 - There is already some `#ifdef BOARD_v1_revA` (for the first prototypes of KBox)
 - If I make a version with the teensy 3.6 later, the test on CPU would not be enough to differentiate your board from mine.

ronzeiller, 2017-12-28:
No problem, I think I can wait until you will do a teensy 3.6 version :-)
(Otherwise I would have to change the platformio.ini every time....)

closed 2017-12-28 by ronzeiller