sarfata/kbox-firmware #122: if Serial 2 input/output both True -> parsing itself

ronzeiller, 2018-01-11:
I do not know, if it makes sense if a Serial port is enabled for input and output at the same time, but if it is than KBox seems to try to parse it´s own sent NMEA again.

**Debug Message:**
```
D SerialService.cpp:162 SerialService[1] Baudrate: 38400 Input: true Output: true
I SerialService.cpp:170 SerialService[2] Baudrate: 4800 Input: true Output: true
D SerialService.cpp:216 Writing NMEA to Serial[1] output: $IIXDR,A,0.9,D,PTCH,A,1.4,D,ROLL*50
D SerialService.cpp:216 Writing NMEA to Serial[1] output: $IIHDM,69.3,M*1E
D SerialService.cpp:216 Writing NMEA to Serial[2] output: $IIXDR,A,0.9,D,PTCH,A,1.4,D,ROLL*50
D SerialService.cpp:183 Serial[2]: Found 1 sentences waiting
D SKNMEAParser.cpp:49 $IIXDR,A,0.9,D,PTCH,A,1.4,D,ROLL*50
D SKNMEAParser.cpp:58 InputNMEA2: IIXDR - Unable to parse sentence
```
(for testing I added a `DEBUG("%s",sentence.c_str());` at line 49)

The weird thing is, that it is does not show up at Serial1.

**btw:** 
```
D SerialService.cpp:162 
I SerialService.cpp:170
```
should be both the same D or I?

sarfata, 2018-01-11:
I have tested this on a kbox hardware when writing the config stuff and I am pretty sure this did not happen. 

Since you also have a problem where nothing shows up on serial2 input I wonder if maybe it is a hardware issue. 

Could the TX and RX lines be shorted together in your hardware? What pins of the 3.6 are you using? How are you doing the conversion to NMEA/RS422 voltages?

sarfata, 2018-01-11:
They should probably be both INFO. They only appear once at start so it’s not too verbose.

ronzeiller, 2018-01-11:
Was checking everything, changing the MAX232, all wires....
Finally I changed the Teensy and voilà it is working. 
Thank You for the hint, that it was hardware related!

closed 2018-01-11 by ronzeiller