sarfata/kbox-firmware #123: Garmin NMEA0183 on Serial input not visible

ronzeiller, 2018-01-11:
@sarfata 
Do you remember, we talked about this some time ago, but I could not find where it was....

The issue is, that NMEA0183 from Garmin are completely ignored.

**The settings are:**
Serial 2 input enabled, Baudrate 4800
direct input of NMEA0183 sentences from Timo's NMEA-Simulator are coming in normal

**Example of Garmin sentences:**
```
$GPBWC,103710,,,,,,T,,M,,N,,D*7A
$GPVTG,2.2,T,358.6,M,0.0,N,0.0,K,D*2E
$GPXTE,A,A,,,N,D*54
$HCHDG,127.0,,,3.6,E*28
$GPRTE,1,1,c,*37
$GPRMC,103713,A,4812.7528,N,01613.6539,E,0.0,2.2,110118,3.6,E,D*1F
$GPRMB,A,,,,,,,,,,,,V,D*19
$GPGGA,103713,4812.7528,N,01613.6539,E,2,12,1.6,228.9,M,43.8,M,,*47
$GPGSA,A,3,02,03,05,06,07,09,16,22,23,26,30,36,2.7,1.6,2.2*38
$GPGSV,3,1,12,02,34,297,17,03,18,126,36,05,06,305,00,06,46,248,24*74
$GPGSV,3,2,12,07,48,185,42,09,82,008,17,16,19,068,22,22,00,127,00*72
$GPGSV,3,3,12,23,50,069,22,26,13,039,00,30,22,199,31,36,32,201,33*7A
$GPGLL,4812.7528,N,01613.6539,E,103713,A,D*4E
$GPBOD,,T,,M,,*47
$GPBWC,103713,,,,,,T,,M,,N,,D*79
$GPVTG,2.2,T,358.6,M,0.0,N,0.0,K,D*2E
$GPXTE,A,A,,,N,D*54
$HCHDG,130.3,,,3.6,E*2D
$GPRTE,1,1,c,*37
$GPRMC,103715,A,4812.7528,N,01613.6539,E,0.0,2.2,110118,3.6,E,D*19
$GPRMB,A,,,,,,,,,,,,V,D*19
$GPGGA,103715,4812.7528,N,01613.6539,E,2,12,1.6,228.9,M,43.8,M,,*41
$GPGSA,A,3,02,03,05,06,07,09,16,22,23,26,30,36,2.7,1.6,2.2*38
$GPGSV,3,1,12,02,34,297,17,03,18,126,36,05,06,305,00,06,46,248,23*73
$GPGSV,3,2,12,07,48,185,42,09,82,008,17,16,19,068,22,22,00,127,00*72
$GPGSV,3,3,12,23,50,069,23,26,13,039,00,30,22,199,31,36,32,201,33*7B
$GPGLL,4812.7528,N,01613.6539,E,103715,A,D*48
$GPBOD,,T,,M,,*47
$GPBWC,103715,,,,,,T,,M,,N,,D*7F
$GPVTG,2.2,T,358.6,M,0.0,N,0.0,K,D*2E
$GPXTE,A,A,,,N,D*54
$HCHDG,130.3,,,3.6,E*2D
$GPRTE,1,1,c,*37
$GPRMC,103717,A,4812.7528,N,01613.6539,E,0.0,2.2,110118,3.6,E,D*1B
$GPRMB,A,,,,,,,,,,,,V,D*19
$GPGGA,103717,4812.7528,N,01613.6539,E,2,12,1.6,228.9,M,43.8,M,,*43
$GPGSA,A,3,02,03,05,06,07,09,16,22,23,26,30,36,2.6,1.6,2.0*3B
$GPGSV,3,1,12,02,34,297,17,03,18,126,36,05,06,305,00,06,46,248,24*74
$GPGSV,3,2,12,07,48,185,42,09,82,008,18,16,19,068,22,22,00,127,20*7F
$GPGSV,3,3,12,23,50,069,23,26,13,039,00,30,22,199,32,36,32,201,34*7F
$GPGLL,4812.7528,N,01613.6539,E,103717,A,D*4A
$GPBOD,,T,,M,,*47
$GPBWC,103717,,,,,,T,,M,,N,,D*7D
$GPVTG,2.2,T,358.6,M,0.0,N,0.0,K,D*2E
$GPXTE,A,A,,,N,D*54
$HCHDG,130.2,,,3.6,E*2C
$GPRTE,1,1,c,*37
```

Working example with Actisense NMEA Reader:
[image]

sarfata, 2018-01-11:
Is the receive counter incrementing? Is the data visible in the logs and then ignored? Or completely not received?

sarfata, 2018-01-11:
How are you connecting the nmea simulator? How are you connecting the Garmin?

ronzeiller, 2018-01-11:
> How are you connecting the nmea simulator?
> How are you connecting the Garmin?

1. NMEA Simulator Win7 -> USB -> USB/Serial FTDI -> KBox Serial 1 or Serial 2 => OK datas are coming in

2. Garmin GPSmap78s with Serial out -> USB/Serial FTDI -> USB Port -> NMEA Reader => datas from GPS OK

3. Garmin GPSmap78s with Serial out -> KBox Serial 1 or Serial 2  => Rx not counting

In KBox I use a normal MAX232 for serial in/out.
It could be that it is because of voltage level from Garmin GPS that it is not recognized.
The only explanation I can think of.

This remembers me, that we should have to have a real USB in/out for datas on KBox......

ronzeiller, 2018-01-11:
Close issue, because I am quite sure that it is Garmin GPS related and not KBox.

closed 2018-01-11 by ronzeiller