sarfata/kbox-firmware #129: Unable to connect to WiFi network using kbox-config.json file

BlackPearl293, 2018-02-18:
KBOX is able to understand that WiFi AP is set as disable (so it reads the config file), but cannot connect to WiFi network:
Display shows WiFi-C Connecting with IP 0.0.0.0

/Johan

kbox-confi.json:

{
  "imu": {
    "enabled": true,
    "frequency": 20,
    "enableHdg": true,
    "enableHeelPitch": true,
    "mounting": "verticalStarboardHull"
  },
  "barometer": {
    "enabled": true,
    "frequency": 1
  },
  "wifi": {
    "enabled": true,
    "nmeaConverter": {
      "xdrPressure": true,
      "xdrAttitude": true,
      "xdrBattery": true,
      "hdm": true,
      "mwv": true,
      "rsa": true
    },
    "accessPoint": {
      "enabled": false,
      "ssid": "KBox"
    },
    "client": {
      "enabled": true,
      "ssid": "BlueCruise",
      "password": "secret33"
    }
  },
  "serial1": {
    "inputMode": "nmea",
    "outputMode": "nmea",
    "baudRate": 38400,
    "nmeaConverter": {
      "xdrPressure": true,
      "xdrAttitude": true,
      "xdrBattery": true,
      "hdm": true,
      "mwv": true,
      "rsa": true
    }
  },
  "serial2": {
    "inputMode": "nmea",
    "outputMode": "nmea",
    "baudRate": 4800,
    "nmeaConverter": {
      "xdrPressure": false,
      "xdrAttitude": false,
      "xdrBattery": false,
      "hdm": true,
      "mwv": true,
      "rsa": true
    }
  },
  "nmea2000": {
    "txEnabled": true,
    "rxEnabled": true
  }
}

sarfata, 2018-02-18:
@BlackPearl293 can you give more information on the network you are connecting to?

Is it 2.4 or 5Ghz? What type of encryption is used?

BlackPearl293, 2018-02-18:
Hi Thomas,

2,4GHz -802.11n.
It is WPA2-personal (my home network) – the unit has no problem to connect when the credentials are hardcoded, but that was ofcourse with an older code version, I just downloaded and installed the new stuff today (expetcting to see the ipaddress onscreen)

Best regards,

Johan Holmqvist

BlackPearl293, 2018-02-18:
Se my comments above
/Johan

BlackPearl293, 2018-07-29:
> @BlackPearl293<https://github.com/BlackPearl293> Did you ever get this working? Looking at this now, I think the problem might be that the wifi module was not the same firmware version as the host.
>
> Did you run both update command?
>
> platformio run -e host -t upload
>
> platformio run -e esp -t upload
>
> So sorry I did not follow up earlier :/

Hi Thomas,

Yes I did both. Will try again with recent updates.

Med vänlig hälsning,

Johan Holmqvist
__________________________________________________________________________
AEC AB
Besöksadress: Första Långgatan 28B (vån4) | Postadress: Box 120 93 | 413 27 Göteborg
Växel: 010-850 20 00 | Fax: 010-850 20 10 | Direkt: 010-850 29 12 | Mobil: 0708-51 16 03
E-post: johl@aec.se<mailto:johl@aec.se> | Hemsida: www.aec.se<http://www.aec.se/>

BlackPearl293, 2018-07-29:
Hi Thomas,

Regarding the kbox-config.json file.

  "imu": \{\par
    "enabled": true,\par
    "frequency": 20,\par
    "enableHdg": true,\par
    "enableHeelPitch": true,\par
    "mounting": "verticalPortHull"\par

What values are valid for “mounting”. I have Starboard Hull (facing port), but I am not sure it is perfectly aligned with the boats longitudal axis.

Also: is there reason to manipulate with frequency?

Med vänlig hälsning,

Johan Holmqvist
__________________________________________________________________________
AEC AB
Besöksadress: Första Långgatan 28B (vån4) | Postadress: Box 120 93 | 413 27 Göteborg
Växel: 010-850 20 00 | Fax: 010-850 20 10 | Direkt: 010-850 29 12 | Mobil: 0708-51 16 03
E-post: johl@aec.se<mailto:johl@aec.se> | Hemsida: www.aec.se<http://www.aec.se/>

sarfata, 2018-07-29:
Johan,

> Now it works, the kbox-config.json file.

Great :) Glad this works!

> However there is a new error message also available as fatal at time of upload on the PC screen ”fatal: not a .git repository (or any of the parent directories): .git”

The git error is because you just downloaded a zip instead of using git to update the source code. This makes it impossible for kbox to detect the current version.

It's not critical, it just means you will not get the version number on the screen.

> What values are valid for “mounting”. I have Starboard Hull (facing port), but I am not sure it is perfectly aligned with the boats longitudal axis.

The correct syntax for your mounting position is `verticalStarboardHull`. There is no option to adjust the axis slightly. 

> Also: is there reason to manipulate with frequency?

Frequency allows you to change how fast the message is sent. With 20, the heel/roll angles will be sent 20 times per second. I would recommend reducing this to 1 or 5 because otherwise it will fill up your logs very fast.

thomas

sarfata, 2018-07-29:
@BlackPearl293 the images are not included in github replies. Can you email it directly to me or upload it in the issue via the github web interface?