sarfata/kbox-firmware #130: KBOX Log File is zero bytes (empty)

BlackPearl293, 2018-02-18:
A log file is created with unique number each boot, but the files are all empty.

sarfata, 2018-02-18:
Do you have anything connected to your KBox via NMEA or NMEA2000?
When did you last update and re-upload KBox?

BlackPearl293, 2018-02-18:
No – today was the new download.

ronzeiller, 2018-02-18:
@sarfata 

> Do you have anything connected to your KBox via NMEA or NMEA2000?

I always asked myself: what was the reason why you do not write the values of internal sensors to the logfile?

BlackPearl293, 2018-02-18:
Things like connecting to network?

ronzeiller, 2018-02-18:
@BlackPearl293   

Yes, things like that and also some important messages as, which instruments found on N2k network, and may be start up informations, which sensor is on and working, and so on....

@sarfata 
Would be a nice thing to have, that main.cpp at start-up and services can send messages which are written to the log-file.

sarfata, 2018-07-27:
Updating tickets ...

The value of internal sensors is now recorded (since v1.3.0 in May) in signalk format in the logs. 

Log messages from the "system" are also recorded if they are not debug messages. So all INFO/ERROR messages are now saved and I have included more information at startup too. 

I think adding the list of NMEA2000 equipment detected is a great idea. We have issue #139 for this.

I will close this ticket for now because there has been tons of changes around logs and I think this is solved but let me know if you still have issues @BlackPearl293.

closed 2018-07-27 by sarfata