sarfata/kbox-firmware #131: How to convert NMEA0183 sentence which needs different SKUpdates?

ronzeiller, 2018-02-26:
Some NMEA0183 need more than one SKUpdate, e.g. the RMC sentence which consists of:

UTC Time
Latitude
Longitude
Speed over ground, knots
Track made good, degrees true
Date, ddmmyy
Magnetic Variation, degrees

But how to get the values?

sarfata, 2018-02-27:
@ronzeiller you can include all of this in one SKUpdate object. One `SKUpdate` can contain multiple values. For example the existing [`parseRMC`](https://github.com/sarfata/kbox-firmware/blob/master/src/common/signalk/SKNMEAParser.cpp#L61) function already includes SOG, COG, Latitude and Longitude.

Does that answer your question?

ronzeiller, 2018-02-27:
@sarfata 
Sorry, I am afraid not.

We get from NMEA2000 different PGN which are in separated `SKUpdate`
1. PGN 129025 Position, Rapid Update Lat/Lon
2. PGN 129026 COG SOG rapid
3. PGN 127258 Magnetic Variation (for this we even do not have SignalK if I remember right...)
4. PGN 126992 UTC Time and Date (We do not have SKUpdate for this as a string is missing)

**All**those  PGNs are needed to convert into **one** NMEA0183 sentence $IIRMC