sarfata/kbox-firmware #132: [Wiki] URL to NMEA2000 and Future PGN Tables

shantanoo-desai, 2018-03-09:
__Supported PGNs__ page provides a false link to the `N2KMessage.h` (HTTP 404)

And due to some Markdown misalignments the Future PGN table is completely misaligned.

I have changed the information locally but need access to change the Wiki Page.

     remote: Permission to sarfata/kbox-firmware.wiki.git denied to shantanoo-desai.
     fatal: unable to access 'https://github.com/sarfata/kbox-firmware.wiki.git/': The requested URL returned error: 403

(Just a follower of the repo, trying to help :grinning: )

sarfata, 2018-03-10:
Thanks @shantanoo-desai - I am not sure how to do a PR on a wiki page but if you just send me the content or a link to it, I will update it and credit you. Thanks a lot!

shantanoo-desai, 2018-03-10:
Hi, just a [setting away](https://help.github.com/articles/changing-access-permissions-for-wikis/)

sarfata, 2018-03-10:
@shantanoo-desai should work now!

shantanoo-desai, 2018-03-10:
Data changed. Thanks.

closed 2018-03-10 by shantanoo-desai

ronzeiller, 2018-03-10:
@shantanoo-desai 

PGN 129026 | COG & SOG | is already supported by KBox

It provides Course Over Ground (COG) and Speed Over Ground (SOG)

~~We should support receiving this to set the time and also sending it when we are connected via NMEA to a GPS~~

Thank You.

shantanoo-desai, 2018-03-10:
@ronzeiller Added the Information in the available PGNs table and striked the old information out

ronzeiller, 2018-03-10:
👍