sarfata/kbox-firmware #149: Add values of barometer sensor to one of the pages

ronzeiller, 2018-12-09:
I did it once on the Voltage-Page, before you introduced the current.

May be barometric pressure in a third line? 
(Could be necessary to move date from Voltage to stat-page to get more space?)

Wanted to discuss with you before I do some coding with a PR

sarfata, 2018-12-09:
I think you should do a separate screen, just the way you like and then we can have a config option to choose which one they want. You can start from the current one, copy/paste it and let's call it "MainStyle2" or something like this.

ronzeiller, 2018-12-10:
Yes, a separate screen with graphics of past 12 or 24 hours would be great!

But as people cannot easily switch through pages (if I understood it right, the switch is under the case) they might want to have as many informations as possible?

This is why I thought of a third line on IMU-Page for barometric pressure for the moment.
And later, when the graphics is developed to transfer it on a separate screen.

For this approach it even would not need an own config for pressure display now.