sarfata/kbox-firmware #150: Do not get NMEA over KBox WiFi

ronzeiller, 2018-12-09:
Hello Thomas,

WiFi access point is working on 192.168.4.1
Connection to WiFi is ok
KBox display shows (1) connected
ping is ok => 64 bytes from 192.168.4.1: icmp_seq=4 ttl=255 time=4.410 ms

But then?
For openCPN TCP/IP or telnet a port is needed
How to get datas over WiFi?

sarfata, 2018-12-10:
> Now I just have to find out again why only NMEA for engine battery is sent (not the supply voltage), when config is like this:

Smells like a bug ...

It _should_ work looking at the code but maybe it does not. Or maybe something bad happens which triggers the error messages. Can you try to disable `xdrBattery` and see if you still get the errors?

Conversion is done in SKNMEAConverter. You could add some logs in there too.
https://github.com/sarfata/kbox-firmware/blob/master/src/common/signalk/SKNMEAConverter.cpp

ronzeiller, 2018-12-10:
disabled xdrBattery in wifi config
no inputs over serial
Actisense NGT-1 on N2k-bus (one PCDIN message at startup)
WiFi as Access Point

After 40 seconds uptime the first entry of error frame:

```
36758;LWI;src/esp/main.cpp:128|ESP Connected clients: 0 Free heap: 36224 Uptime: 36s
39248;LHI;src/common/comms/KommandHandler.cpp:32|Received error frame!
39439;LHI;src/host/os/TaskManager.cpp:77|KBox uptime: 40s RAM Used: 16196 bytes Free: 16276 bytes
```
And then it continued:

```
40216;LHI;src/common/comms/KommandHandler.cpp:32|Received error frame!
41388;LHI;src/common/comms/KommandHandler.cpp:32|Received error frame!
41760;LWI;src/esp/main.cpp:128|ESP Connected clients: 1 Free heap: 33648 Uptime: 41s
42309;LHI;src/common/comms/KommandHandler.cpp:32|Received error frame!
43336;LHI;src/common/comms/KommandHandler.cpp:32|Received error frame!
44358;LHI;src/common/comms/KommandHandler.cpp:32|Received error frame!
```
[kbox-23.log](https://github.com/sarfata/kbox-firmware/files/2664339/kbox-23.log)

**Same, but WiFi connected to my WLAN
Serial debug:**
(Voltages for engine, house and dc are not connected, Teensy pin not at ground => -416.57V)

```
D KommandHandlerWiFiStatus.cpp:53 WiFiStatus: state: 2 dhcpClients: 0 tcpClients: 0 signalkClients: 0 ip: 192.168.1.13
D SerialService.cpp:219 Writing NMEA to Serial[1] output: $IIHDM,45.7,M*14
D SerialService.cpp:219 Writing NMEA to Serial[1] output: $IIXDR,A,4.6,D,PTCH,A,-91.4,D,ROLL*4F
D SerialService.cpp:219 Writing NMEA to Serial[2] output: $IIHDM,45.7,M*14
D SerialService.cpp:219 Writing NMEA to Serial[2] output: $IIXDR,A,4.6,D,PTCH,A,-91.4,D,ROLL*4F
D BarometerService.cpp:45 Read temperature=27.58 C and pressure=98901.8 Pa
D SerialService.cpp:219 Writing NMEA to Serial[1] output: $IIXDR,P,0.98902,B,Barometer*2D
D SerialService.cpp:219 Writing NMEA to Serial[2] output: $IIXDR,P,0.98902,B,Barometer*2D
D KommandHandlerWiFiStatus.cpp:53 WiFiStatus: state: 2 dhcpClients: 0 tcpClients: 0 signalkClients: 0 ip: 192.168.1.13
D SerialService.cpp:219 Writing NMEA to Serial[1] output: $IIXDR,V,-416.57,V,engine*72
D SerialService.cpp:219 Writing NMEA to Serial[1] output: $IIXDR,V,-416.57,V,house*18
D SerialService.cpp:219 Writing NMEA to Serial[1] output: $IIXDR,V,-416.57,V,dc3*48
D SerialService.cpp:219 Writing NMEA to Serial[1] output: $IIXDR,V,12.03,V,kbox-supply*40
D SerialService.cpp:219 Writing NMEA to Serial[2] output: $IIXDR,V,-416.57,V,engine*72
D SerialService.cpp:219 Writing NMEA to Serial[2] output: $IIXDR,V,-416.57,V,house*18
D SerialService.cpp:219 Writing NMEA to Serial[2] output: $IIXDR,V,-416.57,V,dc3*48
D SerialService.cpp:219 Writing NMEA to Serial[2] output: $IIXDR,V,12.03,V,kbox-supply*40
D KommandHandlerWiFiStatus.cpp:53 WiFiStatus: state: 2 dhcpClients: 0 tcpClients: 0 signalkClients: 0 ip: 192.168.1.13
```

**Logged on SD-Card:
Only one error frame at the beginning, then no more.**
```

5500;LHI;src/common/comms/KommandHandler.cpp:32|Received error frame!
5531;LWI;src/esp/main.cpp:237|StationModeConnected
5542;LWI;src/esp/main.cpp:233|DHCP Got IP Address!
6755;LWI;src/esp/main.cpp:128|ESP Connected clients: 0 Free heap: 36400 Uptime: 6s
9428;LHI;src/host/os/TaskManager.cpp:77|KBox uptime: 10s RAM Used: 16236 bytes Free: 16300 bytes
```
[kbox-24.log](https://github.com/sarfata/kbox-firmware/files/2664397/kbox-24.log)

sarfata, 2018-12-10:
Do you have the latest version of the WiFi firmware?

Can you add a debug message here: https://github.com/sarfata/kbox-firmware/blob/master/src/esp/main.cpp#L187

To print:
 - `len`
 - `kr.getKommandIdentifier()`
 - `kr.dataSize()`

For example:

    DEBUG("Error details - len=%u identifier=%u dataSize=%u", len, kr.getKommandIdentifier(), kr.dataSize())

Something like it should work. We should see a debug line for every error. This is in the wifi firmware so you will need to update it.

ronzeiller, 2018-12-11:
> Do you have the latest version of the WiFi firmware?

Using your latest master with teensy36 board

Did a `platformio update` too seems all Up-to-date

```
Platform Espressif 8266
--------
Updating espressif8266                   @ 1.8.0          [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-espotapy                   @ 1.1.2          [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20402.4      [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]

Platform Espressif 8266 (Stage)
--------
Updating espressif8266                   @ ac1e0c8        [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-espotapy                   @ 1.1.2          [Up-to-date]
Updating framework-arduinoespressif8266  @ 47d6f83        [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]
```

> `DEBUG("Error details - len=%u identifier=%u dataSize=%u", len, kr.getKommandIdentifier(), kr.dataSize())`

Access Point enabled, no connection no error

Access Point enabled, 1 client via telnet or openCPN:
`D KommandHandlerWiFiStatus.cpp:53 WiFiStatus: state: 2 dhcpClients: 1 tcpClients: 1 signalkClients: 0 ip: 0.0.0.0`

Errors found:

```
WD main.cpp:188 Error details - len=254 identifier=25964 dataSize=252
WD main.cpp:188 Error details - len=17 identifier=11314 dataSize=15
WD main.cpp:188 Error details - len=16 identifier=8748 dataSize=14

```

closing connections => no errors any more